real-stanford/diffusion_policy #2: Running the diffusion policy colab, subprocess-exited-with-error

DanielTakeshi, 2023-03-24:
I am attempting to run the state-based colab. But the first cell seems to be running into an issue.

[image]

I think the first cell tries to run the installation code, but I get the error as shown in the screenshot above. For your copy/paste convenience, it's this text here:

```
Python 3.9.16
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Is there something that I am missing with regards to how to run and use colab? I am also running into the same error with the vision-based Diffusion Policy colab.

DanielTakeshi, 2023-03-31:
@cheng-chi  Someone else also independently told me of running into this issue. We're wondering if you might be able to take a look? This way we can quickly build upon your work to increase its impact. :)

cheng-chi, 2023-03-31:
Related to https://github.com/openai/gym/issues/3176

AnqiaoLi, 2023-06-12:
I still run into the same issue with `setuptools==65.5.0` . And I fixed it by adding `pip==22`. Which is also mentioned in  [openai/gym#3176](https://github.com/openai/gym/issues/3176)

cheng-chi, 2023-06-13:
@AnqiaoLi Thanks for the suggestion! I just updated the CoLab notebook with pip==22