real-stanford/diffusion_policy #11: Question about the observation

Icamd, 2023-07-02:
Hello, thank you for providing the code! I just started learning about the diffusion model and was deeply inspired after reading your paper. In your paper, you mentioned using observation to guide the CNN base diffusion model for action generation, transform observation Ot into observation embedding sequence by a shared MLP. Is this something you first proposed or an improvement on previous papers? Thank you!

closed 2023-07-04 by Icamd

cheng-chi, 2023-07-06:
Hi @lcamd sorry for my delayed reply. Converting observations into an embedding sequence is a quite common practice in robot learning. Therefore is's difficult to pinpoint which publication proposed it first. My specific implementation is deeply inspired by [robomimic](https://robomimic.github.io/), which you might find useful for citation purposes.

Icamd, 2023-07-06:
@cheng-chi Thank you for your reply! It is common to convert observations into an embedding sequence in robot learning, but your work is the first one to use the embedding sequence in the diffusion model right? And are there any difference between the model in google-colab and the model in the standard code you release on the github? Thank you!

cheng-chi, 2023-07-06:
@lcamd I'm not certain if we are the first but feel free to cite our paper on it! 
In the codebase I implemented 2 variants of the policy architecture (CNN vs Transformer) and 2 variants of observation encoder (Standard ResNet18 with global average pooling vs SpatialSoftmaxPooling borrowed from robomimic "hybrid policy"), in total of 4 variations. The CoLab only implements the CNN + standard ResNet18 variant, but it should be representative of diffusion policy's performance on most datasets.

Icamd, 2023-07-06:
Thank you for your patience! I wish you all the best in your research!