real-stanford/diffusion_policy #15: Question: about the franka's end-effector

AlbertTan404, 2023-08-16:
Thanks for your novel work!

I'm curious about what's the end-effector of the panda robot in the 6DoF Flipping task? Did you print that or that's an official/third party component? I'd appreciate it if you could show me the way.

cheng-chi, 2023-09-07:
Hi @AlbertTan404, the finger is 3D printed using [NinjaFlex TPU](https://ninjatek.com/shop/ninjaflex/). Here is the onshape link for its [CAD model](https://cad.onshape.com/documents/bec64ccbab1c554f3cfea1e3/w/b54dc11dc0673c6df11e275c/e/67993c5ee6c667ded1e4c5be?renderMode=0&uiState=64fa57db3b8b5f4ef163eead).

AlbertTan404, 2023-09-08:
> Hi @AlbertTan404, the finger is 3D printed using [NinjaFlex TPU](https://ninjatek.com/shop/ninjaflex/). Here is the onshape link for its [CAD [...]

Really appreciate for your reply!

closed 2023-09-09 by cheng-chi