real-stanford/diffusion_policy #25: How to generate low dim embeddings the for Kitchen Envs

xavihart, 2023-10-11:
In the dataset, the observation sequence is sized N*60, is there any code snippets to generate the low dim embeddings?

cheng-chi, 2023-10-29:
Hi @xavihart I'm not quite sure I understand your question. The native observation space for the kitchen task is 60 (though a lot of these dimensions are empty), and the dataset is directly extracted from [*.mjl files](https://github.com/google-research/relay-policy-learning/blob/master/kitchen_demos_multitask.zip) published in the original [relay-policy-learining repo](https://github.com/google-research/relay-policy-learning) inside [KitchenMjlLowdimDataset](https://github.com/real-stanford/diffusion_policy/blob/548a52bbb105518058e27bf34dcf90bf6f73681a/diffusion_policy/dataset/kitchen_mjl_lowdim_dataset.py#L37)

xavihart, 2023-10-31:
I got it, so it is not vision-based?

cheng-chi, 2023-10-31:
Yes, we only have state-based version of the kitchen task in this repo.

closed 2023-11-01 by xavihart