real-stanford/diffusion_policy #35: What are the differences between low_dim.hdf5 and low_dim_abs.hdf5?

Junxix, 2023-12-02:
Hi Cheng, I'm wondering what are the differences between low_dim.hdf5 and low_dim_abs.hdf5 since the dataset in the robomimic environment only has low_dim.hdf5 and image.hdf5. How is the low_dim_abs.hdf5 and  image_abs.hdf5 dataset produced?
Thanks for your help!

cheng-chi, 2023-12-11:
Hi @Junxix, in robomimic's offical dataset, all actions are represented as the delta with respect to the current pose. We call this action represetnation "velocity control" in this paper. Unfortunately, CNN-based diffusion policy performs very poorly under this action represetantion. Therefore, we converted all robomimic datasets into absolute posititional actions using [robomimic_dataset_conversion.py](https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/scripts/robomimic_dataset_conversion.py). The result of this conversion is saved to low_dim_abs.hdf5 and image_abs.hdf5 respectively.

deavn2236, 2024-05-31:
The reading method for the diffusion policy dataset is in the form of zarr. Why is it HDF5? Do we need to convert HDF5 to zarr?