real-stanford/diffusion_policy #37: Extending to a new robot

jkim447, 2023-12-11:
Hi,

Really great work. We are trying to extend this codebase to a different robot and with a significantly different learning setup.

In your README, it is mentioned:

Most of our implementations of Dataset uses a combination of [ReplayBuffer](https://github.com/real-stanford/diffusion_policy#replaybuffer) and [SequenceSampler](https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/common/sampler.py) to generate samples. Correctly handling padding at the beginning and the end of each demonstration episode according to To and Ta is important for good performance. Please read our [SequenceSampler](https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/common/sampler.py) before implementing your own sampling method.

Is it really that simple to understand exactly how sequence sampler works? I feel like it's a bit unapproachable and it would take quite some time to really parse the code and understand what's happening. Has anyone else tried to extend the code without adhering to this structure?

I am in the process of doing that, but as I continue, I am altering significantly parts of the training code because again I am not using the existing SequenceSampler, etc. Any pointers would be helpful!

cheng-chi, 2023-12-11:
Hi @jkim447, significant part of SequenceSampler's complexity comes from handeling different types of padding before and after each episode. However, if you are only interesetd in training on data collected in the real world, it is very likely you don't need any padding at all. This is becuase in real world data collection, human demostrators' reaction time will cause the beginning and end of each episode to be static for a few seconds anyway. 
If you ignore the padding issue, Sequence sampler simply selects all possible subsequences of length `horizon` in each episode.

jkim447, 2023-12-12:
Great, thanks a lot for your response. I went through the jupyter notebook to understand a bit and it makes more sense.

One more question:
I've tried to work with the codebase using my own data format (e.g., I just have .jpg imgs for the observations and .csv files for the kinematics) but as I'm going through the codebase I realize I'm breaking things / need to change a lot of the existing code. I feel that the best way forward is to reorganize to the data format as implemented in the original code. And would you agree that this is the best strategy and not try to hack my way through?

cheng-chi, 2023-12-12:
Hi @jkim447, I think converting a set of jpg images and csv into a ReplayBuffer should be fairly straightforward.
[Here](https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/scripts/bet_blockpush_conversion.py) is an example script for converting BET's npy files (lowdim obs only) into a ReplayBuffer. If storage space is not a concern, you can simply treat images just as another key and follow this script's workflow, which will compress images using lossless compressor.
For better image compression, most of my implementation uses Jpeg2k for image compression inside ReplayBuffer. Example scripts can be found [here](https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/scripts/real_dataset_conversion.py). The multi-threaded image copying code is a bit intimidating. However if you can wait, simply writing a for loop to copy all the images into the corresponding zarr array will get you the same result!

jkim447, 2023-12-15:
After poking around a bit more, I ended-up using the Colab implementation. I think the Colab implementation is quite straight-forward -- though, I am using my own dataloader. Hopefully, this should be enough to achieve some state-of-art performance -- correct?

Thanks for the explanations as well, it makes sense!

closed 2024-04-14 by jkim447