real-stanford/diffusion_policy #38: [Question] How to demo in simulation?

haastregt, 2023-12-18:
Hi,

I would like to use this amazing Diffusion Policy method in a custom simulation. As a first step, I am trying to reproduce some of the simulations given in the paper ('lift', 'can', 'square', ...) and understand how this code works to see how I can make an environment for my own simulation.

In the ReadMe, there are instructions on how to train and evaluate simulations, but using already gathered data for the pygame pusht example. Moreover, in the code itself, I can only find demo_real_robot.py (which obviously interfaces with the real robot) and a demo_pusht.py which interfaces with pygame but not with e.g. the robomimic simulations (which would be interesting to see how e.g. visual information is retrieved from the simulation so I can try replicate that).

Maybe I am missing something obvious, but is there any easy way to reproduced one of the simulated examples from the paper, from demonstrating to evaluation? Or would this require some custom code that is not in this repo?

Thank you!

cheng-chi, 2023-12-18:
Hi @haastregt, I'm using the benchmark data that is already provided by robomimic and never collected demonstration myself! For demonstration collection in robomimic please consult their [documentaiton](https://robomimic.github.io/docs/index.html)

haastregt, 2023-12-19:
Thank you for the fast reply! I will check out their examples then :)

closed 2023-12-19 by haastregt