real-stanford/diffusion_policy #40: robomimic_img demo file is too large to download

yolo01826, 2023-12-23:
Dear author:
     非常感谢你的工作。我这几天正在尝试在robomimic环境里面复现diffusion policy，但是我发现你们提供的处理过的robomimic_img演示文件居然有足足78G大小，这需要相当长的时间才能全部下载。为什么这个文件会这么大，我非常诧异，因为我看了一下robomimic官网的演示数据集可能针对每一个任务只有几百MB大小。所以如果我只想在image输入的can(ph)这一项任务上复现的话，应该怎么解决数据集的问题，是只能下载全部的演示数据还是可以下载robomimic官网的数据集经过后处理用于模仿学习。
                                                                                                                                                                           
[image]
谢谢！

closed 2023-12-23 by yolo01826

changhaonan, 2024-04-13:
Did you solve the problem?

yolo01826, 2024-04-14:
i downloaded the whole dataset

---- Replied Message ----
| From | Harvey ***@***.***> |
| Date | 04/14/2024 05:05 |
| To | real-stanford/diffusion_policy ***@***.***> |
| Cc | yolo01826 ***@***.***>,
State ***@***.***> |
| Subject | Re: [real-stanford/diffusion_policy] robomimic_img demo file is too large to download (Issue #40) |

Did you solve the problem?

XPLearner, 2024-05-27:
@yolo01826 效果怎么样

yolo01826, 2024-05-27:
效果不是很好，高维动作空间下训练表现和实际测试表现差异很大

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/27/2024 11:35 |
| To | real-stanford/diffusion_policy ***@***.***> |
| Cc | faded ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [real-stanford/diffusion_policy] robomimic_img demo file is too large to download (Issue #40) |

@yolo01826 效果怎么样