real-stanford/diffusion_policy #52: shapely.errors.TopologicalError: The operation 'GEOSIntersection_r' could not be performed.

MaximaMaSun, 2024-03-05:
Eval PushtImageRunner 1/1:   0%|              | 0/300 [00:00<?, ?it/s][2024-03-05 13:47:41,822][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 210.5008311236281 342.48464083840656 at 210.5008311236281 342.48464083840656
[2024-03-05 13:47:41,822][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 234.99189481952544 314.67771107771085 at 234.99189481952544 314.67771107771085
[2024-03-05 13:47:41,822][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 220.39022562435579 235.97833701476938 at 220.39022562435579 235.97833701476938
[2024-03-05 13:47:41,822][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 178.67475743781893 209.88693535477299 at 178.67475743781893 209.88693535477299
[2024-03-05 13:47:41,822][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 289.14878950291342 266.5939992541135 at 289.14878950291342 266.5939992541135
[2024-03-05 13:47:41,824][shapely.geos][INFO] - Self-intersection at or near point 289.14878950291342 266.5939992541135
[2024-03-05 13:47:41,824][shapely.geos][INFO] - Self-intersection at or near point 234.99189481952544 314.67771107771085
[2024-03-05 13:47:41,824][shapely.geos][INFO] - Self-intersection at or near point 210.5008311236281 342.48464083840656
[2024-03-05 13:47:41,824][shapely.geos][INFO] - Self-intersection at or near point 220.39022562435579 235.97833701476938
[2024-03-05 13:47:41,824][shapely.geos][INFO] - Self-intersection at or near point 178.67475743781893 209.88693535477299
[2024-03-05 13:47:41,825][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 309.8263772698329 215.10372685673218 at 309.8263772698329 215.10372685673218
[2024-03-05 13:47:41,827][shapely.geos][INFO] - Self-intersection at or near point 309.8263772698329 215.10372685673218
[2024-03-05 13:47:41,825][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 291.18848680659426 277.83949209027594 at 291.18848680659426 277.83949209027594
[2024-03-05 13:47:41,827][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 283.61346399000001 286.18896886258517 at 283.61346399000001 286.18896886258517
[2024-03-05 13:47:41,829][shapely.geos][INFO] - Self-intersection at or near point 291.18848680659426 277.83949209027594
[2024-03-05 13:47:41,829][shapely.geos][INFO] - Self-intersection at or near point 283.61346399000001 286.18896886258517
[2024-03-05 13:47:41,835][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 282.46518743654559 322.54902054743388 at 282.46518743654559 322.54902054743388
[2024-03-05 13:47:41,840][shapely.geos][INFO] - Self-intersection at or near point 282.46518743654559 322.54902054743388
[2024-03-05 13:47:41,825][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 268.47390492719433 304.81559802469485 at 268.47390492719433 304.81559802469485
[2024-03-05 13:47:41,841][shapely.geos][INFO] - Self-intersection at or near point 268.47390492719433 304.81559802469485
[2024-03-05 13:47:41,838][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 318.04646451990203 363.62674201727964 at 318.04646451990203 363.62674201727964
[2024-03-05 13:47:41,841][shapely.geos][INFO] - Self-intersection at or near point 318.04646451990203 363.62674201727964
[2024-03-05 13:47:41,858][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 297.62376437258138 216.36702689446531 at 297.62376437258138 216.36702689446531
[2024-03-05 13:47:41,858][shapely.geos][ERROR] - TopologyException: Input geom 1 is invalid: Self-intersection at or near point 228.68062339104259 244.17213369889146 at 228.68062339104259 244.17213369889146
[2024-03-05 13:47:41,859][shapely.geos][INFO] - Self-intersection at or near point 228.68062339104259 244.17213369889146
[2024-03-05 13:47:41,860][shapely.geos][INFO] - Self-intersection at or near point 297.62376437258138 216.36702689446531
ERROR: Received the following error from Worker-13: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b37f0>
ERROR: Shutting down Worker-13.
ERROR: Received the following error from Worker-24: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033c0190>
ERROR: Shutting down Worker-24.
ERROR: Received the following error from Worker-4: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b7580>
ERROR: Shutting down Worker-4.
ERROR: Received the following error from Worker-25: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033bf190>
ERROR: Shutting down Worker-25.
ERROR: Received the following error from Worker-23: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033bf220>
ERROR: Shutting down Worker-23.
ERROR: Received the following error from Worker-21: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b82b0>
ERROR: Shutting down Worker-21.
ERROR: Received the following error from Worker-9: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b62b0>
ERROR: Shutting down Worker-9.
ERROR: Received the following error from Worker-5: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b66d0>
ERROR: Shutting down Worker-5.
ERROR: Received the following error from Worker-7: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033b7970>
ERROR: Shutting down Worker-7.
ERROR: Received the following error from Worker-38: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033c12e0>
ERROR: Shutting down Worker-38.
ERROR: Received the following error from Worker-52: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033c75b0>
ERROR: Shutting down Worker-52.
ERROR: Received the following error from Worker-39: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033c1430>
ERROR: Shutting down Worker-39.
ERROR: Received the following error from Worker-53: TopologicalError: The operation 'GEOSIntersection_r' could not be performed. Likely cause is invalidity of the geometry <shapely.geometry.multipolygon.MultiPolygon object at 0x7f8d033c7700>
ERROR: Shutting down Worker-53.
ERROR: Raising the last exception back to the main process.

MaximaMaSun, 2024-03-05:
Hello, I encountered the following problem after running the training command. Do you have any suggestions?
​

yuweiDu, 2024-03-07:
I encountered the same error

ifanma, 2024-07-27:
it works for me change line 25 in diffusion_policy/env/pusht/pusht_env.py 
to `geom =sg.MultiPolygon(geoms).buffer(0.1)`

closed 2024-08-05 by MaximaMaSun