real-stanford/diffusion_policy #62: Running eval.py "main  workspace = cls(cfg, output_dir=output_dir) TypeError: __init__() got an unexpected keyword argument 'output_dir'?

johnnevins, 2024-04-19:
Appreciate your work and detailed github page, well done! 
Just have a minor question, when I tried to reproduce the "pusht" example following the instruction, it worked out all well. However, when I switch to other examples(I only changed corresponding yaml and ckpt file and names), it always throw out the error below:

~/diffusion_policy$ python eval.py --checkpoint data/epoch=5900-test_mean_score=1.000.ckpt --output_dir data/transport_eval_output --device cuda:0
Traceback (most recent call last):
  File "/home/ubuntu/diffusion_policy/eval.py", line 64, in <module>
    main()
  File "/home/ubuntu/anaconda3/envs/robodiff/lib/python3.9/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/robodiff/lib/python3.9/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/ubuntu/anaconda3/envs/robodiff/lib/python3.9/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ubuntu/anaconda3/envs/robodiff/lib/python3.9/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/ubuntu/diffusion_policy/eval.py", line 34, in main
    workspace = cls(cfg, output_dir=output_dir)
TypeError: __init__() got an unexpected keyword argument 'output_dir'

same error for "can" case, for example. Wonder why and how to fix it?

closed 2024-04-19 by johnnevins

KAI-AVG, 2024-10-07:
你可以将那个 cls = hydra.utils.get_class(cfg._target_)这个实例化的结果输出出来，然后就可以看到通过hydra的实例化之后的类，很有可能是实例化之后的那个类出了问题

ShouZhengAI, 2024-11-03:
@johnnevins  have you solved the question ? I have the same problem, share your solution dear?