real-stanford/diffusion_policy #81: Why we must to use MultiStepWrapper?

CAI23sbP, 2024-05-16:
How are you @cheng-chi ?

I read your paper and code.
But i didn't understand about MultiStepWrapper.
It only worked in Evaluation stage, Could you explain why?

sigmundhh, 2024-05-24:
Hi! As far I see, the MultiStepWrapper allows for the applications of $T_a$ actions. In [this part of the code](https://github.com/real-stanford/diffusion_policy/blob/548a52bbb105518058e27bf34dcf90bf6f73681a/diffusion_policy/gym_util/multistep_wrapper.py#L105), they iterate over the list of actions (which is $T_a$ long), which is what is returned from the predict_action from Diffusion Policy:
```python
def step(self, action):
        """
        actions: (n_action_steps,) + action_shape
        """
        for act in action:
            if len(self.done) > 0 and self.done[-1]:
         ...
```

Hope this helps!

CAI23sbP, 2024-05-27:
Oh i see. 
Thank you! @sigmundhh .