real-stanford/diffusion_policy #83: how to use multiGPU

deavn2236, 2024-05-23:
How can I use multiple GPUs to train a model simultaneously

YY-GX, 2024-06-16:
I have the same question, it'll be quite helpful if this repo could support multi-GPU training considering it takes such a long time to train some environments.

qureshinomaan, 2024-08-28:
I think the UMI repository has provided a way of training multi GPU training. 
https://github.com/real-stanford/universal_manipulation_interface

I was able to make it work with this repository as well. There are only three changes : 
Using the environment in umi instead of the diffusion policy one..

Use this workspace : https://github.com/real-stanford/universal_manipulation_interface/blob/main/diffusion_policy/workspace/train_diffusion_unet_image_workspace.py

And in https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/model/common/module_attr_mixin.py
make the change :  self._dummy_variable = nn.Parameter(requires_grad=False)

accelerate launch --num_processes <ngpus> train.py --config-dir=. --config-name=image_pusht_diffusion_policy_cnn.yaml training.seed=42 training.device=cuda:0 hydra.run.dir='data/outputs/${now:%Y.%m.%d}/${now:%H.%M.%S}_${name}_${task_name}'

ZTX2021, 2025-02-10:
> I think the UMI repository has provided a way of training multi GPU training. https://github.com/real-stanford/universal_manipulation_interface
> 
> I was able to make it work with this repository as well. There are only three changes : Using the environment in umi instead of the diffusion policy one..
> 
> Use this workspace : https://github.com/real-stanford/universal_manipulation_interface/blob/main/diffusion_policy/workspace/train_diffusion_unet_image_workspace.py
> 
> And in https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/model/common/module_attr_mixin.py make the change : self._dummy_variable = nn.Parameter(requires_grad=False)
> 
> accelerate launch --num_processes train.py --config-dir=. --config-name=image_pusht_diffusion_policy_cnn.yaml training.seed=42 training.device=cuda:0 hydra.run.dir='data/outputs/${now:%Y.%m.%d}/${now:%H.%M.%S}_${name}_${task_name}'

This doesn't work for me. And the config `image_pusht_diffusion_policy_cnn.yaml` seems to use train_diffusion_unet_hybrid_workspace.py instead of train_diffusion_unet_image_workspace.py