real-stanford/diffusion_policy #92: DOES DP REALLY WORK ON REAL TASK?

huangxu1991, 2024-06-19:
We use aloha to make real data on real task like Grasping Cube into Bowl.
We collected about 200 episodes, each episode about 500 steps.
We use simple net config: jsut resnet18 for image, MLP for qpos, unet1d for dolicy, and DDPM noise scheduler from diffusers with default config from colab demo (https://colab.research.google.com/drive/18GIHeOQ5DyjMN8iIRZL2EKZ0745NLIpg?usp=sharing).
We trained 1000 epochs with all four views images. Sadly, it DO NOT WORK AT ALL. 
PS: obs horizon 2, predtion action 16, action horizon 8
@cheng-chi @pointW hi guys, Would you please give some advices?

huangxu1991, 2024-08-02:
it works.

closed 2024-08-02 by huangxu1991

yanrihong, 2024-11-27:
hey bro, what did u do

XinnWang, 2024-12-03:
@huangxu1991 what did u do?

huangxu1991, 2024-12-10:
sorry for late reply. in fact we did a lot in the last few months, mostly debugging which we did not notice before. in fact, DP really works. NO MAGIC indeed.