real-stanford/diffusion_policy #93: Why action dimension of DP is 10 but action dimension of lstm-gmm is 7 in tool_hang task? What's the difference?

Vhyrsd, 2024-06-25:
In tool hang task, the action dimension of Diffusion Policy is 10, different from the one of LSTM-GMM (it's 7). It's the same task and why are the action dimensions are different?

Vhyrsd, 2024-07-01:
@cheng-chi Could you please assist with this?

bkpcoding, 2024-07-07:
The 7 action dimension is for relative position between the two positions that is how the robomimic dataset is collected. Whereas the action dimension is the absolute position of the robot, they use it because they find that DP performs better with abs position (or absolute configuration of the robot) compared to relative position.

wudier483, 2024-10-10:
> The 7 action dimension is for relative position between the two positions that is how the robomimic dataset is collected. Whereas the action dimension is the absolute position of the robot, they use [...]

@bkpcoding  Why does absolute position need to be represented using 10 dimensions, and what are they included in? Looking forward to your response？