real-stanford/diffusion_policy #94: Clarification on Global Condition vs. Local Condition in Diffusion Policy

Gear420, 2024-06-25:
I am currently exploring the diffusion_policy library and I am interested in understanding the distinctions and applications of global condition and local condition within this framework.

What constitutes a global condition versus a local condition in the context of trajectory prediction and decision-making tasks?
How are these conditions processed and utilized differently in the library's architecture?
Are there specific examples or use cases in the codebase where global condition and local condition are distinctly applied?

RochMollero, 2024-07-12:
Same here: how is it related to 'impating' which is the term used in the paper, which is not defined. in the code there seem 3 types of inputs:

input
local_cond
global_cond

i kinda think i nderstand what they are but some confirmation would be great

a-nooj, 2024-07-29:
Here's my understanding -

Inpainting = hardcoding parts of the action to whatever your want. Eg: overwriting the start and goal states after each diffusion step to match your planning problem

local_cond = adding "supplementary information" to the training that is specific to each timestep in the action. Eg: collision cost for a single robot waypoint in the whole trajectory

global_cond = adding supplementary information that applies to the action trajectory as a whole. Eg: RGB image that gives an overview of the scene or language input that tells the robot what to do generally ("pick up the red cup")