real-stanford/diffusion_policy #96: full_key: task.dataset

coderpracticer, 2024-07-25:
Has anyone meets the same error?

Error executing job with overrides: []
Error locating target 'diffusion_policy.dataset.pusht_image_dataset.PushTImageDataset', set env var HYDRA_FULL_ERROR=1 to see chained exception.
full_key: task.dataset

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

LemonQC, 2024-09-26:
Have you solved？

geyan21, 2024-10-10:
Just installing all required packages as in yaml file. The locating error means hydra cannot import the target correctly. You can check by `"from diffusion_policy.dataset.pusht_image_dataset import PushTImageDataset"`. It will probably show some module missing errors.

ShouZhengAI, 2024-11-02:
hello @coderpracticer ,have you tackle this question? can you share with me? it's great important for me. if so ,thank you very muck

ShouZhengAI, 2024-11-02:
@LemonQC  Have you solved？i need your help

ShouZhengAI, 2024-11-02:
> Just installing all required packages as in yaml file. The locating error means hydra cannot import the target correctly. You can check by `"from diffusion_policy.dataset.pusht_image_dataset import PushTImageDataset"`. It will probably show some module missing errors.

it's not used for me, could you see my errors? 

```
(robodiff) asrock@asrock:~/quanbao/ShouZheng/DP$ python train.py --config-dir=. --config-name=image_pusht_diffusion_policy_cnn.yaml training.seed=42 training.device=cuda:0 hydra.run.dir='data/outputs/${now:%Y.%m.%d}/${now:%H.%M.%S}_${name}_${task_name}'
/home/asrock/.local/lib/python3.9/site-packages/torchvision/datapoints/__init__.py:12: UserWarning: The torchvision.datapoints and torchvision.transforms.v2 namespaces are still Beta. While we do not expect major breaking changes, some APIs may still change according to user feedback. Please submit any feedback you may have in this issue: https://github.com/pytorch/vision/issues/6753, and you can also check out https://github.com/pytorch/vision/issues/7319 to learn more about the APIs that we suspect might involve future changes. You can silence this warning by calling torchvision.disable_beta_transforms_warning().
  warnings.warn(_BETA_TRANSFORMS_WARNING)
/home/asrock/.local/lib/python3.9/site-packages/torchvision/transforms/v2/__init__.py:54: UserWarning: The torchvision.datapoints and torchvision.transforms.v2 namespaces are still Beta. While we do not expect major breaking changes, some APIs may still change according to user feedback. Please submit any feedback you may have in this issue: https://github.com/pytorch/vision/issues/6753, and you can also check out https://github.com/pytorch/vision/issues/7319 to learn more about the APIs that we suspect might involve future changes. You can silence this warning by calling torchvision.disable_beta_transforms_warning().
  warnings.warn(_BETA_TRANSFORMS_WARNING)

============= Initialized Observation Utils with Obs Spec =============

using obs modality: low_dim with keys: ['agent_pos']
using obs modality: rgb with keys: ['image']
using obs modality: depth with keys: []
using obs modality: scan with keys: []
/home/asrock/.local/lib/python3.9/site-packages/torchvision/models/_utils.py:208: UserWarning: The parameter 'pretrained' is deprecated since 0.13 and may be removed in the future, please use 'weights' instead.
  warnings.warn(
/home/asrock/.local/lib/python3.9/site-packages/torchvision/models/_utils.py:223: UserWarning: Arguments other than a weight enum or `None` for 'weights' are deprecated since 0.13 and may be removed in the future. The current behavior is equivalent to passing `weights=None`.
  warnings.warn(msg)
[2024-11-02 21:29:23,839][diffusion_policy.model.diffusion.conditional_unet1d][INFO] - number of parameters: 2.515119e+08
Diffusion params: 2.515119e+08
Vision params: 1.119709e+07
Error executing job with overrides: ['training.seed=42', 'training.device=cuda:0']
Error locating target 'diffusion_policy.env_runner.pusht_image_runner.PushTImageRunner', set env var HYDRA_FULL_ERROR=1 to see chained exception.
full_key: task.env_runner

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

```
all the instructions are copied 😂,and input step by step followed with official

Cocatric, 2025-01-08:
I have the same issue with you. Do you have solved it?

Cocatric, 2025-01-08:
> it's not used for me, could you see my errors?
>
> ```
> (robodiff) asrock@asrock:~/quanbao/ShouZheng/DP$ python train.py --config-dir=. --config-name=image_pusht_diffusion_policy_cnn.yaml [...]

I have the same issue with you. Do you have solved it?

quas-modo, 2025-01-27:
You may create __init__.py under diffusion_policy/dataset
```python
from diffusion_policy.dataset.pusht_image_dataset import PushTImageDataset
__all__ = ['PushTImageDataset']
```

humphreyde, 2025-04-23:
may be due to missing dataset or incorrect path