real-stanford/diffusion_policy #102: ImportError: ImportError: cannot import name 'IGNORE_GIT_FOLDER_PATTERNS' from 'huggingface_hub.utils' ....

CodingCatMountain, 2024-08-14:
Awesome Work! Diffusion-policy is very inspired! 

However, could you help me to fix this problem? I try to run state-based ipynb file in colab. I met such error:
`ImportError: cannot import name 'IGNORE_GIT_FOLDER_PATTERNS' from 'huggingface_hub.utils' (/usr/local/lib/python3.10/dist-packages/huggingface_hub/utils/__init__.py)
`
Looking forward to your reply.

mayeRobot, 2024-09-20:
Hello, I tried downgrade the version of hugglingface_hub to 0.22.1 and it works. You can try to execute the following commands.
`pip uninstall -y huggingface_hub`
`pip install huggingface_hub==0.22.1`
Good luck to you!

kevinqyh0827, 2024-11-07:
Hi, I guess this is because "IGNORE_GIT_FOLDER_PATTERNS" is a constant variable in hugginface utils repo and it has been removed/ renamed by the latest version. I found there is a "DEFAULT_IGNORE_PATTERNS" varaible from here: ("https://github.com/huggingface/huggingface_hub/blob/0c98fbd5840ce9dc2ac59745afe079babeb9f353/src/huggingface_hub/utils/_paths.py#L24"). It seems this variable has similar meaning of the previous one. Replace the old "IGNORE_GIT_FOLDER_PATTERNS" with this new one works for me now.