real-stanford/diffusion_policy #109: What is the meaning of the content of eval_log.json?

Shi-jiayu, 2024-08-30:
such as, 
```
{
  "test/mean_score": 0.9,
  "test/sim_max_reward_100000": 1.0,
  "test/sim_max_reward_100001": 1.0,
  "test/sim_max_reward_100002": 1.0,
  "test/sim_max_reward_100003": 0.0,
  "test/sim_max_reward_100004": 1.0,
  "test/sim_max_reward_100005": 1.0,
  "test/sim_max_reward_100006": 1.0,
  "test/sim_max_reward_100007": 1.0,
  "test/sim_max_reward_100008": 1.0,
  "test/sim_max_reward_100009": 1.0,
  "test/sim_max_reward_100010": 1.0,
  "test/sim_max_reward_100011": 1.0,
  "test/sim_max_reward_100012": 0.0,
  "test/sim_max_reward_100013": 1.0,
  "test/sim_max_reward_100014": 1.0,
  "test/sim_max_reward_100015": 1.0,
  "test/sim_max_reward_100016": 1.0,
  "test/sim_max_reward_100017": 1.0,
  "test/sim_max_reward_100018": 1.0,
  "test/sim_max_reward_100019": 1.0,
  "test/sim_max_reward_100020": 1.0,
  "test/sim_max_reward_100021": 1.0,
  "test/sim_max_reward_100022": 1.0,
  "test/sim_max_reward_100023": 1.0,
  "test/sim_max_reward_100024": 1.0,
  "test/sim_max_reward_100025": 1.0,
  "test/sim_max_reward_100026": 1.0,
  "test/sim_max_reward_100027": 1.0,
  "test/sim_max_reward_100028": 0.0,
  "test/sim_max_reward_100029": 0.0,
  "test/sim_max_reward_100030": 1.0,
  "test/sim_max_reward_100031": 1.0,
  "test/sim_max_reward_100032": 1.0,
  "test/sim_max_reward_100033": 1.0,
  "test/sim_max_reward_100034": 1.0,
  "test/sim_max_reward_100035": 1.0,
  "test/sim_max_reward_100036": 1.0,
  "test/sim_max_reward_100037": 1.0,
  "test/sim_max_reward_100038": 1.0,
  "test/sim_max_reward_100039": 1.0,
  "test/sim_max_reward_100040": 1.0,
  "test/sim_max_reward_100041": 1.0,
  "test/sim_max_reward_100042": 1.0,
  "test/sim_max_reward_100043": 0.0,
  "test/sim_max_reward_100044": 1.0,
  "test/sim_max_reward_100045": 1.0,
  "test/sim_max_reward_100046": 1.0,
  "test/sim_max_reward_100047": 1.0,
  "test/sim_max_reward_100048": 1.0,
  "test/sim_max_reward_100049": 1.0,
  "test/sim_video_100000": "results/robomimic/low_dim/can_mh/media/qxdb76fx.mp4",
  "test/sim_video_100001": "results/robomimic/low_dim/can_mh/media/0mr05828.mp4",
  "test/sim_video_100002": "results/robomimic/low_dim/can_mh/media/bxitrahz.mp4",
  "test/sim_video_100003": "results/robomimic/low_dim/can_mh/media/929ab7tf.mp4",
  "train/mean_score": 1.0,
  "train/sim_max_reward_0": 1.0,
  "train/sim_max_reward_1": 1.0,
  "train/sim_max_reward_2": 1.0,
  "train/sim_max_reward_3": 1.0,
  "train/sim_max_reward_4": 1.0,
  "train/sim_max_reward_5": 1.0,
  "train/sim_video_0": "results/robomimic/low_dim/can_mh/media/n3luxxhx.mp4",
  "train/sim_video_1": "results/robomimic/low_dim/can_mh/media/i49viczs.mp4"
}
```
Among them, there are 6 mp4 files, and I noticed that 2 of them are marked as' train' and the other 4 are marked as' training', what's the meaning of these?

Shi-jiayu, 2024-08-30:
And, what is the meaning of the number indexs, such as 10000,10001,...?

abcdsaltfish, 2024-09-08:
*It's "test", not "training".

*10000,10001 are seeds of the env.