real-stanford/diffusion_policy #116: ConnectionResetError(104) in KitchenLowdimRunner during environment initialization

mr20010112, 2024-10-16:
Thank you for the great work on this project. I’ve encountered an issue when running `KitchenLowdimRunner` during environment initialization. Specifically, I’m seeing a `ConnectionResetError` that causes the process to fail when trying to initialize the `AsyncVectorEnv`.

Here is a summary of the problem:

- **Error**: `ConnectionResetError(104, 'Connection reset by peer')`
- **Relevant file**: `diffusion_policy/env_runner/kitchen_lowdim_runner.py`
- **Error stack trace**:
  ```plaintext
  ConnectionResetError: [Errno 104] Connection reset by peer
    File "/home/mrq/project/diffusion_policy-main/diffusion_policy/gym_util/async_vector_env.py", line 387, in _check_observation_spaces
      same_spaces, successes = zip(*[pipe.recv() for pipe in self.parent_pipes])
  ...
  hydra.errors.InstantiationException: Error in call to target 'diffusion_policy.env_runner.kitchen_lowdim_runner.KitchenLowdimRunner':
  ConnectionResetError(104, 'Connection reset by peer')
  ```

The error seems to occur when `AsyncVectorEnv` is initialized, and the observation space check is performed via communication with child processes (`pipe.recv()`), resulting in the connection being reset.

closed 2024-10-20 by mr20010112

ShouZhengAI, 2024-11-03:
@mr20010112  hi bro , have you solved it?

mr20010112, 2024-11-03:
Yes, by setting the gym version to 0.21.0 I guess.

ShouZhengAI, 2024-11-03:
@mr20010112 thx,bro,I will try it

Luca-Wiehe, 2024-11-04:
> Yes, by setting the gym version to 0.21.0 I guess.

Unfortunately, this doesn't change the issue for me. Have you changed anything else additionally?

mr20010112, 2024-11-05:
> > Yes, by setting the gym version to 0.21.0 I guess.
> 
> Unfortunately, this doesn't solve the issue for me. Have you changed anything else additionally? Can you tell us your python version?

https://drive.google.com/file/d/1yhYW4pBkRQlkUtFwvHIGS3H4J--HMkva/view?usp=sharing

mr20010112, 2024-11-05:
> > Yes, by setting the gym version to 0.21.0 I guess.
> 
> Unfortunately, this doesn't solve the issue for me. Have you changed anything else additionally? Can you tell us your python version?

3.9.15

weleen, 2025-05-13:
> > Yes, by setting the gym version to 0.21.0 I guess.
> 
> Unfortunately, this doesn't solve the issue for me. Have you changed anything else additionally? Can you tell us your python version?
@Luca-Wiehe @mr20010112 
Hi, have you solved the issue? I got the same error with python==3.9.15 and gym==0.21.0.