real-stanford/diffusion_policy #130: Robomimic Version Incompatability

Benjamin-So, 2024-12-08:
Thank you for sharing such great work!

**TLDR:**
How can I visualize trajectories for all demos in the Robomimic transportation task with enlarged dimensions? Using Robomimic's built-in script to render videos, I encounter errors due to differences in XML metadata between Diffusion Policy and Robomimic. How were the hdf5 files preprocessed for Diffusion Policy's version of the Robomimic transport task, as the backgrounds also differ?

**Full Description**
I wanted to see if it would be possible to visualize the trajectories for all demos in the transportation task but with enlarged dimensions. For example, I tried stitching the image observations in the hdf5 files, but using this method I can only generate videos with 84x84 dimensions (in the Robomimic transportation case). Is there a clean way to achieve higher resolution?

Robomimic provides a built-in script for executing actions stored in an hdf5 file within their environment and saving the resulting video. The script is described here: [Robomimic Dataset Contents](https://robomimic.github.io/docs/tutorials/dataset_contents.html).

Using this script, I ran:
`python playback_dataset.py --dataset /pathto/diffusion_policy/data/robomimic/datasets/transport/mh/image.hdf5 --use-actions --render --render_image_names shouldercamera0_image
`

However, I encountered an error due to differences in the structure of the XML metadata for the hdf5 transport task provided by Diffusion Policy. This raises a key question:

How were the hdf5 files preprocessed for the Robomimic tasks in Diffusion Policy? I also noticed differences in the background when visualizing image observations for the transport task in Diffusion Policy compared to the original Robomimic dataset.

Would appreciate any insights or guidance!

---
I obtained the video below by executing the actions provided in the transport demo_0 hdf5 file downloaded from the robomimic website. I used the command provided by robomimic to rollout the trajectory.

https://github.com/user-attachments/assets/36228504-0488-4a53-bd5d-facc591dad39

---

I obtained the video below by stitching together the observations in the image transport demo_0 task:

https://github.com/user-attachments/assets/e7582e91-fb0d-4f28-9fcc-26add4178962

mr20010112, 2024-12-26:
Hello,

Could you kindly share the requirements.txt file for this environment? I’ve been unable to run it successfully.

Here’s an example of the path where the file might be located: data/outputs/2024.12.14/16.29.56_train_diffusion_transformer_lowdim_kitchen_lowdim/wandb/latest-run/files/requirements.txt

Thank you in advance for your assistance!

humphreyde, 2025-05-14:
I can use it, which version of robosuite and robomimic do you use ?