real-stanford/diffusion_policy #132: AssertionError in Fixed Workers in Robomimic

mr20010112, 2024-12-22:
### **Description:**

I am encountering a recurring issue with parallel environment execution in `AsyncVectorEnv` in Robomimic. Specifically, certain fixed workers consistently fail during execution, raising `AssertionError`. Below are the details of the issue:

---

### **Environment Details:**

- **Python version:** `3.9.15`
- **Robomimic version:** `0.3.0`
- **Gym version:** `0.21.0`
- **Torch version:** `1.12.1`
- **Other relevant dependencies:** 
    - **Robomimic:** `0.3.0`
    - **Robosuite:** `1.4.1`
    - **Gym:** `0.21.0`
    - **Mujoco_Py:** `2.1.2.14`
    - **Mujoco:** `3.1.6`
    - **D4RL:** `1.1`
    - **dm-control:** `1.0.20`
    - **dm-env:** `1.6
    - **mjrl:** `1.0.0`

---

### **Steps to Reproduce:**

1. Initialize an `AsyncVectorEnv` with multiple environments (28 environments) in Robomimic.
2. Start parallel execution.
3. Fixed workers (e.g., Worker-0, Worker-1, Worker-6, Worker-7, Worker-8) consistently raise `AssertionError`.

---

### **Observed Behavior:**

The following error is encountered:
```
ERROR: Received the following error from Worker-0: AssertionError: 
ERROR: Shutting down Worker-0.
ERROR: Received the following error from Worker-1: AssertionError: 
ERROR: Shutting down Worker-1.
ERROR: Received the following error from Worker-6: AssertionError: 
ERROR: Shutting down Worker-6.
ERROR: Raising the last exception back to the main process.
```

Additionally, the issue appears to be isolated to these specific workers during execution. Debugging logs indicate that the issue may occur during `step` or `reset` calls in the environments.

---

### **Additional Information:**

   ```
1. Debug logs for a single worker:
   ```
   Worker-0: Initializing environment.
   Worker-0: Received command `step`
   Worker-0: Error during command `step`: AssertionError
   Worker-0: Environment closed.
   ```

2. The code snippet for the error-prone logic:
   ```python
   def _raise_if_errors(self, successes):
       if all(successes):
           return
       num_errors = self.num_envs - sum(successes)
       assert num_errors > 0
       for _ in range(num_errors):
           index, exctype, value = self.error_queue.get()
           print(f"Worker-{index} encountered error: {exctype.__name__}, {value}")
           self.parent_pipes[index].close()
           self.parent_pipes[index] = None
       raise exctype(value)
   ```

---

closed 2025-01-01 by mr20010112

White-oranges, 2025-02-16:
I have the same problem. Have you solved it yet?

EDGE-qy, 2025-04-28:
I have the same problem. Have you solved it yet?