real-stanford/diffusion_policy #137: Is it necessary to get another abs dataset or can I use my image. hdf5 as well?

carloskiyoto, 2025-01-24:
Hi, thanks for all the work, I have a question
I'm changing my workspace to TrainDiffusionTransformerHybridWorkspace and is required an abs dataset ( image_abs.hdf5).
Can I import my image_hdf5 or do I have to find a abs dataset?
If I have to find another one, where do I get the abs format? I got the image_hdf5 of robomimic documentation ( https://robomimic.github.io/docs/datasets/robosuite.html )

jangyicheng, 2025-05-21:
Have you solved it?

jangyicheng, 2025-05-23:
The solution is: 

`cd diffusion_policy/diffusion_policy/scripts/`

and locate the robomimic_dataset_conversion.py file, and execute it using the following command:  

`python robomimic_dataset_conversion.py -i /path_in/your_file.hdf5 -o /path_out/your_file.hdf5 -e ./eval`