real-stanford/diffusion_policy #145: cannot import name 'cached_download' from 'huggingface_hub'

Cup-Laboratory, 2025-04-15:
在运行  python train.py --config-dir=. --config-name=image_pusht_diffusion_policy_cnn.yaml training.seed=42 training.device=cuda:0 hydra.run.dir='data/outputs/${now:%Y.%m.%d}/${now:%H.%M.%S}_${name}_${task_name}'
出现了报错情况，大概是 cannot import name 'cached_download' from 'huggingface_hub'
但是我检查了我虚拟环境中的pip 里面是有这个包的，但是没有在yaml里找到 你们能运行成功吗？有大佬教教如何跑通吗？

CarolBaggins2023, 2025-04-16:
In new version of 'huggingface_hub', 'cached_download' seems to be removed. Try downgrade huggingface_hub, such as 0.11.1.

Cup-Laboratory, 2025-04-16:
ok，thank u,  did you succeed in replicating this paper.Can you build this simulation?

CarolBaggins2023, 2025-04-17:
> ok，thank u,&nbsp; did you succeed in replicating this paper.Can you build this simulation?
> […](#)

I have not encountered any problem now. I just follow the README file, Installation:Simulation, Reproducing Simulation Benchmark Results:Download Training Data and Running for a single seed.

Cup-Laboratory, 2025-04-17:
ok. If I have any other questions, I'll ask you for advice.

  
ok，thank u,  did you succeed in replicating this paper.Can you build this simulation?
 …

sakshi79, 2025-06-09:
> ok，thank u,&nbsp; did you succeed in replicating this paper.Can you build this simulation?
> […](#)

I downgraded to huggingface_hub==0.25.0, and was able to replicate results for various tasks, including PushT and Kitchen ones